Stu142/FreeCAD-Gridfinity-Workbench #1: Request to add parameter(s) to change part placement

dragon-architect, 2024-06-10:
When the Gridfinity workbench generates a part, it does so with one of the grid units (with respect to the X and Y axes, it is the front-left corner unit) centered on the origin. The generated blank cannot be moved unless it is part of a body, but then it will be offset by -20.75mm from the origin of the body's coordinate system on both the X and Y axes. If one desires to work from the front-left corner of a generated blank as a coordinate origin, one would need to compensate for this by changing the attachment offset of the geometry sketches (refer to attached screenshot).

Would it be possible to add a parameter or a set of parameters that can be set to change the placement of generated blanks using a multiple of gridfinity grid units?

In the implementation that I have in mind, positionX=0 and positionY=0 positions the generated blank entirely within the positive X and Y axes of the global coordinate system, with the 0.25mm tolerance also included. positionX=-0.5 and positionY=-0.5 could therefore replicate the current default hard-coded placement of generated gridfinity blanks. For a 2x2 gridfinity blank, positionX=-1 and positionY=-1 would center the blank entirely around the global coordinate origin.

[image]

Stu142, 2024-06-12:
This is a great idea, it should be fairly easy to implement the starting position to be 0,0 instead of offset by half a grid.

It should also be easy to add a movement variable as you describe. 

I'm not sure when I will get to this I'm quite busy at the moment but when I do it shouldn't take long.

dragon-architect, 2024-06-16:
I look forward to seeing it implemented when you do get around to it! Thanks for taking the idea. :)

Stu142, 2024-06-24:
@bcwhite-code the z placement is actually an intentional choice so if you go to make a custom cuts into the bin blank you can set your sketch to be on the XY plane. 

This avoids attaching the sketch to a face where you could run into Topological naming problems. 

You could offset the sketch from the XY plane alternatively but that is more complicated for a beginner.

I could potentially make it so the other bins generate at z0 at the bottom if there was reason to. 

Thanks for asking though I'm always open to ideas.

bcwhite-code, 2024-06-25:
Yeah, that makes sense.  I just need to find an easy way to make holes to the maximum depth, keeping 2mm of baseplate.

BTW, this is a really useful addon and works well.  Thanks!

Stu142, 2024-06-27:
You can reference the properties of the bin in dimensions of sketches or features l.

So for example you draw a sketch on top of a blank bin then pad and put the expression 

BinBlank.TotalHeight - 2mm 

This will give you the two 2 mm at the bottom 

Same this applies to all properties, I have been meaning to add some documentation on this to the page.

bcwhite-code, 2024-06-28:
Nice.  I didn't know about that.  I've been attaching the sketch to the bottom, moving it -2mm Z then doing a pocket "through all".

Ruudjhuu, 2025-01-29:
I would like to comment on the X/Y location of objects with a more programming point of view.

I think we should standardize that all functions (Also the functions attached to the buttons in the GUI) should return any kind of shape (3D or 2D) with the origin in the middle. Currently this is not the case. From a programmer point of view, I don't want to find out how the object is positioned if I call a function. The object is needed in the current scope. If the object needs to be somewhere specific, the current scope should be responsible for this translation. The current implementation should not depend on the internals of another function. It would make the life of a programmer easier if I can assume that any object I want to use has it's origin in the middle.

Stu142, 2025-01-29:
@Ruudjhuu Are you talking about the end result or just the individual generation functions?

I would like the final result to be with the bin positioned as in the image below. 

[image]

This allows easier customization, such as with the blank bin and creating a custom cutout after generating the bin. If you decide that you want the bin to be one unit longer it means the whole bin doesn't move on you just the outer edge. This effect is even stronger if you are making a parametric cutout that references parameters of the bin. it would be possible to do but In my opinion easier and more user friendly with the starting position as above. 

I have started implementing the above locations in PR#55 as I had to go through all the functions anyway. 

How it gets to that point, I am wide open for recommendations but pretty set on the final location.

Ruudjhuu, 2025-01-30:
I was talking about the end result and individual generation functions. It seems you already have thought about it. I could  argue that having the axis in the middle would help a lot with any features that would be symmetrical. If I draw a part of a custom cutout and want to mirror this, I now need to first create a datum plane in the middle of the bin to make it work.

I also tend to draw my sketches using symmetry. For example starting with a midpoint rectangle, draw 1/4th of a sketch and use polar pattern. This might be our difference in ways of working which I think is fine. Both ways have plus and downsides in different scenarios. 

Programmatically I would still like to advocate for the origin to be in the middle. Because we generally want to have as few lines of code, patterns like mirror and polar should be used often. It could reduce the lines of code already with a single square (2 lines + 1 polar pattern vs 4 lines of the square). You could see this in the baseplate middle cutout which I touched recently. By using 2 times mirroring, the function was redused by 60-ish lines. 

Regarding you having the need to touch a lot of functions when changing locations. I think the responsibility of the location of objects lays too deep in the code. For example, the implementation for placing features in an array (anything grid-like) exists in almost every function separated from each other. This is a refactor step I might pick up in the future.

Stu142, 2025-01-30:
@Ruudjhuu, you make a very good point here, enough to justify a parameter in the bin to change the location to center on the origin based on what you are doing. 

However, I have thought about this quite a bit in the past and would like the default location to be the 0,0 one as above. 

In addition to my previously mentioned points I have seen the workbench used in an assembly setting and having the bins grow in all directions can cause complications there as well. For someone who is new to FreeCAD or CAD in general, I think having some sides fixed could be more intuitive and cause less headaches but I am really just guessing here to be honest. 

There are clear benefits to both approaches and it is hard to pick a clear winner. 

For individual feature generation I am open to suggestions. The way I am setting it up currently is the feature is generated then moved the correct position before being returned to the bin that is calling it. This would allow a centered approach during the generation of the individual features, then translate to the bin position. I saw the mirrors that you added to the baseplate cut out and that was a great idea to simplify the code in that section, I will definitely be keeping that method in mind. 

There is definitely room for more standardization between the features. I have started to implement some structure changes you proposed in #29 and it is making a huge difference having the properties in just one place.

Ruudjhuu, 2025-01-30:
Your assembly argument is a good one, as assembly features are not yet that mature in FreeCAD it will help to have something fixed. I like these kinds of discussions to get to the best possible solution. 

Regarding pr #29, this would be a big improvement in clean and clear code. Unfortunately it is going to be a very big change. Most of my earlier PRs were to prepare for #29. As I need to make myself familiar with the code base and need it to be more easy to read. I might do some more refactoring before I pick up #29 again, but we'll see what is needed. If you are moving forward with the idea in #29 please make a draft pr so we won't do exactly the same thing.

Stu142, 2025-01-30:
I have taken heavy inspiration from PR#29 in #55 which is currently in draft and does include large changes. It will take a little while to get that completely done, and I am sure there will be room for improvement. 

I was not super familiar with the methods used so in some places I kind of just "made it work" not knowing what the best or proper method was. There will be some cleaning up to do but I think it is at least moving in the right direction and will be easier moving forward.

greg19, 2025-03-18:
What is the status of this issue? There is `GenerationLocation` param for all shapes, is there anything else intended to be implemented?

Stu142, 2025-03-18:
I have been considering adding a `xLocationOffset` and `yLocationOffset` as when a bin is generated withing a part design body the Attachement Offset property is no longer available and it is not possible to move the bin within the body. 

I am not totally sure if this in necessary however the point of the workbench in my opinion is ultimate customization and I believe that using it within a Part Design Body is a common use case so I am leaning towards adding these. 

Alternativly maybe there is a way to show the Attachement properties when generating the bin.

Stu142, 2025-03-18:
@greg19 I discovered that the `Placement` property is still created the same as everything else, just hidden for this scenario, but can be shown with `obj.setEditorMode("Placement", 0)` so easy fix to this after all.