Stu142/FreeCAD-Gridfinity-Workbench #2: Trying to close a sketch on the top blank bin surface crashes FreeCAD or give a "recompute failed" error

mjskier, 2024-06-24:
This looks like a very useful workbench. However I'm not sure how to use it.
My naive try (in FreeCAD 0.21.2 on Linux)

- start in part design
- create body
- switch to Gridfinity workbench
- click Blank Bin, maybe change one of the grid unit
- go back to part design
- click on the top surface
- create a sketch
- create a circle or a rectangle
- close the sketch

At this point FreeCAD either crashes, or I get a "Recompute failed" before I do anything else (like try a pocket)3
This is with the latest version of Gridfinity I just downloaded

mjskier, 2024-06-24:
Never mind. I just watched the gif tutorial. The sketch needs to be on the XY plane, not on the top surface of the bin.

closed 2024-06-24 by mjskier

Stu142, 2024-06-27:
Definitely using the XY plane is the way to go that's why I put it there to avoid attaching the sketch to a face and running into the topological naming problem, 

However, I tested it out of curiosity and it worked fine for me both ways so not sure what is going on here but if you have any more info that might help pinpoint the problem I would like it to work both ways.

mjskier, 2024-06-30:
This only happens on my Linux Mint system.
I tried the same experiment on a MacBook with no issue.

Interestingly enough, on the Mac I was able to install the workbench from the Tools->Addon manager.
On the Linux system it didn't show up on the Addon Manager. So I manually downloaded it to my share/freecad/Mod folder. Once a blank bin is added to a body, any update of the model display is extremely slow.

I'm not sure what prevents a workbench from showing up in the addon manager. Maybe an incompatible version of Python?
Does Freecad use its own Python, or does it depends on the system version?