Stu142/FreeCAD-Gridfinity-Workbench #4: Object of type ViewProviderDocumentObject is not JSON serializable

mjskier, 2024-08-03:
On version 0.21.2 of FreeVAD, on both macos and linux, I get this traceback every time I save the document.
I also get it randomly, which I assume is when freeCAD auto saves.

This doesn't seem to break anything, as the document saves fine and I'm able to export it to 3d print it.
But obviously it pollutes the report view, and makes it hard to see potentially serious errors.

`
09:38:27  PropertyPythonObject::toString(): failed for <class 'freecad.gridfinity_workbench.commands.ViewProviderGridfinity'>
09:38:27  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable

`

Stu142, 2024-08-04:
I unfortunately do not have a macOS or Linux machine to test on. I don't currently know what might be causing this but I will keep it in mind.

mjskier, 2024-08-04:
I'll try to look into it. That's a good excuse for me to try to understand FreeCAD internals.
Since it occurs when saving, I assume FreeCAD tries to serialize the object so it can write it out, and either it, or some object contained inside it is missing a function.
Kind of odd it doesn't happen on Windows, but maybe the FreeCAD Python on Windows is packaged differently and has defaults that are not available in Linux/macOS

FlashSystems, 2024-09-05:
Using Gridfinity Workbench Version 0.5.0 and FreeCAD 0.21.2 I could reproduce this problem on Windows as well:

```
19:41:13  PropertyPythonObject::toString(): failed for <class 'freecad.gridfinity_workbench.commands.ViewProviderGridfinity'>
19:41:13  Traceback (most recent call last):
  File "E:\Program Files\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "E:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "E:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "E:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```

Stu142, 2024-09-07:
It appears the section of code it is referencing I copied from the FreeCAD gear workbench and don't really know the ins and outs of it. All of the view related stuff. 

I'll have a bit of a closer look and keep it in mind but unfortunately I just don't know, sorry. 

Thanks for following up and maybe someone else might be able to weigh in at some point.

mjskier, 2024-09-07:
Probably related to this. (Looks specific to v0.21.2)
https://forum.freecad.org/viewtopic.php?t=86423

mjskier, 2024-09-11:
The traceback is gone with FreeCAD 1.0 RC1
So I think this issue can be closed as a regression specific to v0.21.2

Stu142, 2024-09-15:
Thanks for following up, glad to hear it works in 1.0

closed 2024-09-15 by Stu142