Stu142/FreeCAD-Gridfinity-Workbench #5: gridfinity object slows FreeCAD considerably

mjskier, 2024-08-03:
This is more of a question than a bug report.

Once I get a gridfinity "Blank Bin" in my document FreeCAD uses a lot of resources, slowing everything down. To the point that adding fillets once I'm happy with whatever I designed essentially locks FreeCAD with 100% CPU consumption.

My machine has 40G or RAM and is a somewhat older but still potent I7 8 core CPU. So it shouldn't be too wimpy to run FreeCAD correctly.

I'm not sure if the gridfinity object is just very complex and causes very extensive recomputes every things changes, or if there is something else going on. But for example, double clicking on the BinBlank object in the model tree to get access to the properties, just changing the height, or x or y Grid Units gets very sluggish.

I'm just curious if that is a known issue, and if there are ways to work around the problem.
I'm willing to help to get this understood. (For example, are there any debug variables in the code that can be turned on to get an understanding of what is going on?)

Stu142, 2024-08-04:
What happens for me when creating a bin the compute time can be up to 10 seconds for a larger bin like 5X5 after each change, but after that its fine when just looking at it. My understanding is that FreeCAD recomputes everything after each change, whereas most paid CAD packages keep track and only recompute the changed parts which can lead to long compute times in FreeCAD. There are a fair number of features in the bins. 

You can toggle on skip recomputes by right clicking on the document in the feature tree, its one of the options. then you can change multiple things then recompute manually. 

however, my computer does exhibit this slow behaviour or high usage just long recomputes not sure what it could be unfortunately. Running an 8th gen intel I5 and 3080, 16gb ram.

mjskier, 2024-08-04:
Thanks for the info. I was afraid of that.
I'm still going to use it, I love that workbench. Thanks for taking the time to put it together.

closed 2024-08-04 by mjskier