Stu142/FreeCAD-Gridfinity-Workbench #7: Allow custom grid size on all parts

blessedbiped, 2024-08-22:
The custom grid size is available in the baseplate, but not editable in the other parts.
Ideas: 
- the "zz Expert Only" menu can be collapsed by default and read only switch dropped or 
- store parameters in the spreadsheet workbench for the ability to edit/export per project

Stu142, 2024-08-22:
Thanks for pointing that out, it's an easy fix for me to make the custom grid editable for all the parts. I'll look into making it collapsible that could be nice. 

As for the second suggestion I don't really understand, the way it should currently work is once you generate a bin you can save it as a FreeCAD file and then open that up with any changes you made previously being saved and all properties are always editable. 

As an additional side note the spreadsheet workbench causes a big slowdown when working with complex parts so I will be staying away from that but am open to other suggestions for the functionality you are looking for.

Stu142, 2024-08-23:
Custom Grid Size is now unlocked on all Bins. 

I am going to look further into better ways to organize the properties in the future.

Stu142, 2024-08-23:
please let me know if you find any bugs related to custom grid size I tested it but not thoroughly

closed 2024-08-23 by Stu142