Stu142/FreeCAD-Gridfinity-Workbench #10: Allow custom divider height

mikenorgate, 2024-09-06:
It would be great to be able set a custom height for the x and y dividers independently from the total bin height. Having the divider lower can help being able to grab parts that a closely stacked together

In this example, being able to have the y dividers at half height would allow me to get my fingers in to grab the parts in the bin
[image]

Stu142, 2024-09-07:
Great idea, I've got it mostly working and will be included with the V0.6.0 release probably in the next couple days.

Stu142, 2024-09-10:
implemented in new release

closed 2024-09-10 by Stu142

mikenorgate, 2024-09-10:
Tried this out today, works great. Thanks for your work