Stu142/FreeCAD-Gridfinity-Workbench #11: Label holder is a tiny little bit too high.

JelleHissink, 2024-09-13:
When I use the generator, and I stack another box on top, the bottom of the box actually slides somewhat below the level of the label.

So in fact the box is resting on the holder of the label, especially if you would put a sticker on there. If I make a stack of boxes of 1x1, the stack actually leans over because it rests on top of the lip.

Maybe lower the lip with the label 0.3 mm or so? 

This issue is also observed with boxes that were generated by Zack, so maybe the issue is somewhere else?

Stu142, 2024-09-14:
I've seen this mentioned before as well, and I agree lowering it slightly would be the fix. I'll add it to the list for the next release shouldn't be too hard to implement.

Thanks for the suggestion.

Stu142, 2025-01-05:
fix implemented into v0.7.0

closed 2025-01-05 by Stu142