Stu142/FreeCAD-Gridfinity-Workbench #13: Errors saving

mhr320, 2024-09-23:
Hello. Excellent Workbench. Thank you so much. I am using it daily. 
I have noticed after creating the workbench in Freecad 0.21.2 and maybe in the previous release as well, but can't remember for sure, I am receiving errors from Python that "Object of type ViewProvider is not JSON serializable. I have included a screenshot and bought 3 coffee's for your troubles. Again, thank you for making this workbench. I don't know enough yet about Freecad to know if this is an issue with my install. I am using Debian. I will say regardless of the errors, I am able to save in Freecad and export my .stl files without further problems. The errors occur only after I create the gridfinity object. 
[image]

mhr320, 2024-09-23:
Thanks for getting back to me and I wish I could donate more :). I will keep an eye out for the update.

Stu142, 2025-01-12:
It seems like this is no longer an issue so I am going to close this one, but please feel free to open another if you do encounter it.

closed 2025-01-12 by Stu142