Stu142/FreeCAD-Gridfinity-Workbench #14: Support decimal fractions for bin sizes

gnestlinger, 2024-09-24:
With a grid unit of 42mm, often a lot of drawer space is wasted. Therefore, I like to create my bins in sizes of multiples of 0.5 gridfinity units. Consequently, also the grid on the boittom of the bins should be in multiples of 0.5 units.

The OpenSCAD Gridfinity generator can do that, but since I wanted to create a bin witch custom compartments, I wanted to use this FreeCad implementation, which, unfortuanetly, lacks this feature.

Stu142, 2024-09-24:
I believe I have enabled the grid size parameter so you can set the grid size to be 21 mm and I think it should work but I have not thoroughly tested it.

I plan to add the functionality you are describing just haven't gotten to it yet. I also plan to add more information about this to the readme on how to do all this just haven't gotten there. 

If the custom grid size parameter can't be edited you can change it manually from the const.py file in the workbench location and it should scale the features accordingly. 

If you are looking to still have magnets in the corners that won't currently work but I will consider adding that when I implement the full functionality. 

Hopefully it works for you, let me know if it doesn't.

gnestlinger, 2024-09-27:
Thanks, tuning the grid size parameter worked!

Ruudjhuu, 2025-01-29:
@Stu142 I think this issue can be closed

closed 2025-01-30 by Stu142