Stu142/FreeCAD-Gridfinity-Workbench #15: One magnet hole per gridfinity bin feet

ucool321, 2024-10-04:
Hi

I am using your freecad workbench (version 0.6). I am tried to print only a quarter size bin (20mmx20mm) but the quantity of magnet holes are "hardcoded"? I want only one hole per feet (centric). Is there something wrong configurated or is this feature missing?

[image]

Stu142, 2024-10-04:
Currently, this feature does not exist but I am planning to add it soon.

When you say centric do you mean in the center of each quater, becuase that would not be the gridfinity standard and wopuld not work with most baseplates. The spec is 7.75mm from the outside of the bin to the center of the magnet hole. Is there a specific baseplate that you use for this?

Building on that if someone wanted to use an odd number of half grids with the magnets in the standard location. the bin magnets would not line up with the baseplate magnets in all positions. 

If you have any thoughts on these issues please let me know I want to have the best implementation possible.

Stu142, 2024-10-04:
Potentially I could add an option for both ways of doing it, just don't to get carried away with parameters and cause confusion

ucool321, 2024-10-05:
Thanks for the answer. I am not so familiar with programming.

With centric i mean: If you draw diagonal line from 2 corner points on the feet and take from the line the middle point. 
The option to choose between the gridfinity spec (4 magnets) or one magnet for non spec bins sounds good.