Stu142/FreeCAD-Gridfinity-Workbench #16: Open sided bin option?

phil-barrett, 2024-10-06:
Great workbench! Makes gridfinity easy to get started on. 

I like the gridfinity system but the size of my (and a lot of other) 3D printer limits the size that you can do. With the grid, the screw option allows multiple to be joined together. But, I'm limited to 6x6 module size on the bins (actually even smaller than that as my printer - Bambu X1C - has problems with one corner).  It would be great to be able to make a bin with an open side or two.  Then I could put them together to get larger bin sizes. Maybe add some sort of joiner mechanism like a spline you print separately.

Anyway, I'm sure you get lots of suggestions so won't hold my breath.

Phil

Stu142, 2024-10-07:
Thanks for the Suggestion, It is a good idea for sure but not at the top of my priority list. I might get to it at some point I'll add it to my feature list. 

A couple questions about the implementation that you are looking for. Do you want the bin to be structurally solid with the two halves connected or would it just need a bit of a connection so it can sit in a drawer and never needs to be moved. . 

If there is a certain method of joining bins that you have seen sharing a photo would be great.

phil-barrett, 2024-10-07:
Yeah, I kind of figured it wasn't a top priority.  No problem.

It doesn't need to be structurally solid.  Mostly for larger items in a tool drawer. The pieces probably don't need to joined but a simple joiner spline might be easy.  I don't have a photo of a joining method but was thinking 2 slots, say 5 mm wide by maybe 10 mm long that would align between two halves to allow a 5x20 spline to fit in. 

Have you see the [Clough42 video](https://www.youtube.com/watch?v=RYA0xLryF-g&t=5s) on his take on Gridfinity.  He has some interesting ideas. Not sure how many of them translate to your workbench, though.

By the way, one other idea that might be interesting to you is a "padding" piece.  No drawers in my house and shop have interior dimensions that are a multiple of 42mm.  I've resorted to printing little blocks that make the grid fit snuggly.  Not at all hard for me to do for simple ones that just sit there but a little trickier if using screws to hold the grids together.

Anyway, again, thanks for doing this.  it is very useful!  Great job.

Phil

Stu142, 2024-10-07:
Okay, thanks for the info.

Padding stuff is much higher on the list. Hoping to have additional blocks but also be able to extend the bins and baseplates by an odd amount on end.

phil-barrett, 2024-10-07:
I'm looking forward to that!

greg19, 2025-02-02:
I want to add that you can achieve this easily using Constructive Solid Geometry in FreeCAD. This is the power of having a Gridfinity in an actual CAD software instead of a simple STL generator.

1. In Gridfinity Workbench, create the bin you want
2. Switch to Part Workbench
3. Create a cube, and change it dimensions and position so it encompasses the part of the bin you want to print
4. Select both the part and the cube in Model panel
5. Click Intersection in the toolbox

You can also make the cube encompass the part of the bin you want to cut out, and use Cut instead of Intersection

An example:
[image]