Stu142/FreeCAD-Gridfinity-Workbench #17: Custom shape generation

mjack3k, 2024-10-17:
Hi, love the workbench, big thanks for that. 

I was thinking, that it would be nice, if we could have some more "intelligent" base generation.

For example: 
- we measure the dimensions of our drawer, space on the desk, floor, etc - whatever we want to use for our storage
- we create a sketch that mimics the area
- we tell the generator to "create baseplate(s) that fit in the area"
- generator will create base plates, as long as full cell can fit

With additional parameter "Max part size", where se say the maximum grid to generate - so it wont create one single piece with 30x30 cells, only split it to printable parts that fit on the bed. 

I know that in most situations we talk about rectangular area, but lets assume you have a cabinet under your sink - those sometimes have drawers with a cutout in the middle, for the pipes. 

Similar for the bins - I can think of scenarios, where having a non-rectangular shape would be needed, and a custom shape, maybe like on of the Tetris blocks, would work better and allow for more efficient use of the drawer - for example, a glue gun could fit in an L-shaped bin. 

[image]

Stu142, 2024-10-17:
Thanks for the interesting ideas,

I think having more bin and baseplate shape options would be really cool such as an L as you suggested or U, and this should be doable. No timeline but I'll add this to my new feature list. 

As for the custom shape with a sketch input that is a very interesting idea but I don't know how that could be implemented at this time but I'll think about it. Please share if you have any suggestions.

Stu142, 2025-03-18:
Custom bin and baseplates have been implemented. As for the bigger picture idea of generating the whole area I don't think that is within the scope of the workbench for me. I feel its strength is being able to add customizations onto the bins and baseplates. This idea could pair nicely with a potentential online calculator of sorts to determine what parts are needed which could then be generated and customized in the workbench.

closed 2025-03-18 by Stu142