Stu142/FreeCAD-Gridfinity-Workbench #18: Placing bins on the grid

mjack3k, 2024-10-19:
So I am trying to use this Workbench to create custom containers for drawers in my bathroom.
I would like to recreate the whole drawer and place the various bins in right places, so I can have nice overview of what is where - maybe to help me manage upgrades in the future (maybe to add smaller containers underneath the current ones, for that I need to see which containers can be moved up). 

So the general idea is:
- I select a container/bin (which was created with Gridfinity WB)
- I press "Move along the grid/Place on the grid" button (unsure how to call it) on the toolbar
- It asks me to select the base plate
- I click on one of the baseplates I created with Gridfinity WB
And here is where the fun happens - it knows the bin grid size (for example, 2x3), creates a highlighted rectangle of this size, and align it to the cell my mouse cursor is hovering on.
This way we could place the bins aligned to the cells more easily

Hope that what I described makes sense to anyone. The idea with a simple, highlighted rectangle is to avoid dragging real object (less cpu intensive, I think? Some of us work on lower-end machines). 

Unsure how hard would it be to implement it, but I think all the objects created with Gridfinity WB can have some extra properties and know their origin (it sure looks like, when I change grid size, it always grows in known direction), so it is just a matter of aligning the two. 
I would love to help on this one aswell, but it looks more complicated than I thought.

Stu142, 2024-10-19:
I don't think I really have the knowledge to implement something like this at the moment but I am loving the Ideas and we will see what happens in the future. 

In the meantime, I saw a video someone made recently using the Ondsel assembly workbench with the assembly workbench [https://www.youtube.com/watch?v=tuHAhKb8qz8](https://www.youtube.com/watch?v=tuHAhKb8qz8)

It gets some of the way there. There is a built in assembly workbench in the upcoming FreeCAD 1.0 release as well that should work similarly.

greg19, 2025-02-24:
This can be achiever by using the built in *Transform*, by setting *Translation Increment* to the grid size:

[image]

### Some inconveniences:

#### 1. Bins below baseplate

The default bin generation generates the bins under the baseplate, so they first have to be transformed (with different *Translation Increment*) to a sensible Y position.

From the discussion regarding this at #1:

> I could potentially make it so the other bins generate at z0 at the bottom if there was reason to.

I think this issue is *a* reason, but not a very pressing one. I think adding additional parameter is **not** a good solution and it shouldn't be implemented this way. Neither is merging it with *Generation Location* parameter (which in my opinion should be deleted).

> You could offset the sketch from the XY plane alternatively but that is more complicated for a beginner.

As a CAD beginner I just entered the *Sketcher* for the first time (not looking at any tutorials, just exploring the buttons in FreeCAD) and it was natural for me to first click on the face I want to draw on, then click *Create sketch*. The pop-out selected *Plane face* by default, so I just clicked *OK* and started drawing, easily creating my first pocket without relying on the positioning of the object.

#### 2. Z axis

You cannot specify a separate Z *Translation Increment*, so you if you want to place bins on top of another you have to change the *Translation Increment* back and forth between grid size and height unit.

This will be solved in FreeCAD 1.1, as *Transform* will have its interface extended:

greg19, 2025-02-24:
This can be achiever by using the built in *Transform*, by setting *Translation Increment* to the grid size:

[image]

### Some inconveniences:

#### 1. Bins below baseplate

The default bin generation generates the bins under the baseplate, so they first have to be transformed (with different *Translation Increment*) to a sensible Y position.

From the discussion regarding this at #1:

> I could potentially make it so the other bins generate at z0 at the bottom if there was reason to.

I think this issue is *a* reason, but not a very pressing one. I think adding additional parameter is **not** a good solution and it shouldn't be implemented this way. Neither is merging it with *Generation Location* parameter (which in my opinion should be deleted).

> You could offset the sketch from the XY plane alternatively but that is more complicated for a beginner.

As a CAD beginner I just entered the *Sketcher* for the first time (not looking at any tutorials, just exploring the buttons in FreeCAD) and it was natural for me to first click on the face I want to draw on, then click *Create sketch*. The pop-out selected *Plane face* by default, so I just clicked *OK* and started drawing, easily creating my first pocket without relying on the positioning of the object.

I think it would be better to generate the bins so that they align with baseplates by default.

#### 2. Z axis increment

You cannot specify a separate Z *Translation Increment*, so you if you want to place bins on top of another you have to change the *Translation Increment* back and forth between grid size and height unit.

IIUC, this will be solved in FreeCAD 1.1, as *Transform* will have its interface extended:

[image]

#### Changing *Translation increment*

If there would be a possibility to change the default *Translation Increment* on a per-object basis, it would be great to set it to the grid size (from the object parameters), but I don't know if that is possible.