Stu142/FreeCAD-Gridfinity-Workbench #19: Add a way to set default parameters

mjack3k, 2024-10-19:
It would be nice if we could somehow set our starting parameters, so when we create bins, we don't have to switch everything.

Personally, I always set the bases to be very simple (no screw holes, no magnet holes), and usually would like to go for 3x3 instead of 2x2. This means I need to change those properties for each bin. 
The size isn't as important, since I do work on different sizes, but I usually would like to have same base for all the bins in the project.

Stu142, 2024-10-19:
Now this is doable. you can navigate to the file location User\AppData\Roaming\FreeCAD\Mod\FreeCAD-Gridfinity-Workbench\freecad\gridfinity_workbench\const.py

open the const.py file

Changing these values is effectively changing the defaults of the workbench. Currently not all parameters are here but I can add that pretty easily per bin even so that will come in the next update. I'll add some documentation on how to do this to the main page for the future. 

Note that the values will get overridden with workbench updates but it shouldn't be too hard to change them back. 

From a quick google there dones't seem to be an easy way to add a workbench settings page or something like that so I think this is the best we have got.

Stu142, 2024-10-19:
actually alternatively you can save a file with a generated bin and use that as your template per say instead of generating a new one but I think it would be good to add the default thing anyway

mjack3k, 2024-10-19:
Okay, great, I will definitely look into this file.