Stu142/FreeCAD-Gridfinity-Workbench #20: Make modules grow "up" when changing the height

mjack3k, 2024-10-19:
When changing the height of the bins, they grow towards the bottom (Z-).

The bin needs to be then repositioned again in the grid. 

If possible, it would make my life much easier if they would grow upwards - if there are any good reasons for the current behavior, then maybe have the option to easily change it in the python files somewhere?

[image]

[image]

mjack3k, 2024-10-19:
Also, sorry for spamming with the issues. I just think this is a great workbench and I would love to see it grow.

Stu142, 2024-10-19:
There is a reason for this, it's so it's easier to sketch custom cutouts of the blank bin from the top. Making it possible to use the XY plane and avoid the TNP when making these sketches. 

I never really anticipated this use case so it might change in the future but for now, it is likely to stay the same

Stu142, 2024-10-19:
Actually I have a partial work around. Go to the placement property of bin and put TotalHeight -4.25mm. 

This will then automatically adjust the placement from then on with the caveat that it doesn't seem to take effect right away. you change the height then change another parameter and it will automatically adjust. 

Being able to reference the bin properties in dimensions is the coolest part of the workbench in my opinion. 

[image]

[image]

Stu142, 2024-10-19:
Further Update: 

if you use the forumla HeightUnits* 7mm -4.25 it works perfectly

[image]

mjack3k, 2024-10-19:
Okay, got it.

Stu142, 2025-01-12:
I appreciate the suggestion but I have decided to leave things the way they are.

closed 2025-01-12 by Stu142

mjack3k, 2025-01-12:
Fair enough, in the meantime I got used to this.