Stu142/FreeCAD-Gridfinity-Workbench #21: Feature Request: Seperate outer wall hight from inner space hight

neroes, 2024-11-18:
it would be great if we could seperate the inner wall hight from the outer
then we can make things like screwdriver inserts that are stackable

Stu142, 2024-11-19:
I'm not sure I understand what you mean. Does this apply to a specific type of bin or all of them

neroes, 2024-11-19:
sorry for not being more specific, i mean the blank bin, which i would use to make things such as slotted insert spaces, but currently the bin kinda expects the surface to be the stacking surface, so if the item i make a slot for is ½ below the surface ½, above it interferes with the stack ability

i recently made a holder for my TS101, that looks like the picture, but as you can see i can't put another bin on top
for this it is fine since this is intended to be at the top, but i imagine making more of these where that isn't the case
[image]

Stu142, 2024-11-20:
Yup, I get it now good idea!

No timeline, unfortunately, but I'll update you when I get around to it.

closed 2025-02-07 by Stu142