Stu142/FreeCAD-Gridfinity-Workbench #22: Support for FreeCad 1.0.0

vvranjek, 2024-11-22:
Addon manager can't find the Gridfinity workbench.

Stu142, 2024-11-22:
Try refreshing the local cache in the bottom left corner of the addon manager. 

Other than that, it still seems to be on there as far as I can tell, but I'm not super knowledgeable about the git side of things.

mongrelion, 2024-12-02:
I just installed the workbench on FreeCAD 1.0.0 and it's working as expected.
@vvranjek make sure that you have added the **Custom repository** in `Preferences -> Addon Manager -> Addon manager options`.

The following values worked for me:

**Repository URL**: `https://github.com/Stu142/FreeCAD-Gridfinity-Workbench`
**Branch**: `master`

@Stu142 I think you can close this issue.

vvranjek, 2024-12-03:
I installed it without a problem now too. Not sure if what was the reason before or if there was a fix, but works as intended. We can close the issue. Before we do, my experience with this workbench is hta it is very slow. It takes 3-10 seconds for the workbench to process any changes to gridfinity parameters. It was the same on older versions. I'm on Ubuntu 24.04. Is it just me or is it slow for all of us? Often my OS even thinks that freecad stooped responding because it takes so long.

mongrelion, 2024-12-03:
@vvranjek that sounds like an unrelated issue. Please close the issue and feel free to open up a new one where we can discuss performance. Make sure you mention your specs (Ubuntu 24.04, plus what's your CPU, your GPU if any, how much ram, etc.).

Stu142, 2024-12-03:
@vvranjek That's just how it is with the delay, unfortunately, although I haven't seen more than 3 seconds on Windows, usually less. i5 8400 and 3080.

closed 2024-12-03 by Stu142