Stu142/FreeCAD-Gridfinity-Workbench #25: Error creating Dividers

begrie, 2024-11-26:
I like the workbench - thank you for creating and sharing.

Using FreeCad 1.0 I have detected a reproducible error when creating dividers for EcoBins:
x Grid Units 2
y Grid Units 5
y Dividers 4
-> pyException: Traceback (most recent call last):
  File "C:\Users\Bernd\AppData\Roaming\FreeCAD\Mod\Gridfinity\.\freecad\gridfinity_workbench\features.py", line 72, in execute
    gridfinity_shape = self.generate_gridfinity_shape(fp)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Bernd\AppData\Roaming\FreeCAD\Mod\Gridfinity\.\freecad\gridfinity_workbench\features.py", line 549, in generate_gridfinity_shape
    compartements = MakeEcoBinCut(self, obj)
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Bernd\AppData\Roaming\FreeCAD\Mod\Gridfinity\.\freecad\gridfinity_workbench\feature_construction.py", line 932, in MakeEcoBinCut
    func_fuse = func_fuse.makeFillet(obj.InsideFilletRadius, b_edges)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: class StdFail_NotDone BRep_API: command not done
12:09:20  EcoBin: class StdFail_NotDone BRep_API: command not done

[image]

Stu142, 2024-11-28:
Thanks for letting me know, it seems to only affect that specific combination of y Grid Units and Y dividers and I wasn't able to track down the problem right away. I'll take a harder look at it when I get a chance.

Ruudjhuu, 2024-12-14:
I did some digging. 

It seems that the 4th generated divider is in an unlucky position where the adjoined edge in the bottom corner is `1.48mm` and FreeCad fails to generate a fillet of `1.5mm`. Just `0.02mm` short.

[image]

A workaround would be to put `Inside Fillet Radius` on `1.48mm` to get the maximum inner fillet.

As for a structural solution, this is not that easy. Depending on the location of the wall, the edge in question could be very small or not even touching the corner. This is an artifact of the `Economical Cutout`. 

Some options:
1. Check if the vertexes of the to be filleted edges are bigger than `Inside Fillet Radius`. If a connected edge is smaller use the biggest possible fillet. This probably need an improvement on the edge filtering. Currently all edges from the outer wall are also part of the fillet input but ignored by the fillet function. 

2. Just try to fillet, if it fails decrease `Inside Fillet Radius` until it works. Easy to implement, but kind of a stupid solution.

3. Try applying the fillet once, print warning if it fails, proceed without fillet.

4. Build123D has a [max fillet](https://build123d.readthedocs.io/en/latest/direct_api_reference.html#topology.Mixin3D.max_fillet) function. This seems to be some variant on solution one and resource heavy. I don't know if FreeCad also support this kind of functionality (they both use OCCT kernel)

@Stu142 What are your thoughts?

closed 2025-03-18 by Stu142