Stu142/FreeCAD-Gridfinity-Workbench #46: Bug: Python Dependancy notification on Installation

Stu142, 2025-01-10:
Putting this here to document the issue. Working on a fix. 

When installing the workbench through the addon manager, the following notification is received for requirements that are not required to use the workbench, just for development. 

[image]

I removed the requirements.txt file in version 0.7.1 to see if that fixes it.

Stu142, 2025-01-10:
@Ruudjhuu I revoved the requirements.txt file as it was causing the above prompt during install from the FreeCAD add on manager and I if I am understanding correctly those requirements only apply for development not for use.  

does removing reqirements.txt cause issues for anything else you set up? just wanted to check.

Stu142, 2025-01-10:
Also, just for clarification, everything works fine if you ignore the message it's just confusing for users.

Ruudjhuu, 2025-01-10:
I was not aware FreeCAD uses the `requirements.txt` file.

Removing it breaks the GitHub actions so all future PRs are blocked. Maybe for next time also for small changes you make, create a PR to check integrity. It is usually advised to protect the master branch (GitHub has protection rules).

I'll try to migrate the `requirements.txt` to `pyproject.toml` which support optional dependencies.

I do not yet know how to tests installing this workbench with the add-on manager while not having a new release.

Stu142, 2025-01-10:
It makes sense to do everything through PR, and I'll make sure to do that in the future, thanks for the tip. 

As for testing, I don't know either, but I am happy to just try it and then revert it quickly if it doesn't work.

closed 2025-01-11 by Stu142

Ruudjhuu, 2025-01-12:
I think this issue closed automatically with the PR, but we have not validated it yet, right?

Stu142, 2025-01-12:
@Ruudjhuu Yeah it was closed early, this morning I was able to verify that things are working properly with no warning during install. So it is fixed now.

Thanks for fixing that.

PR #47 (2025-01-11): Fix dependencies
fixes #46

I think this should work. However, I am not sure how to test this in the add-on manager without a release as I also wrote in #46.