Stu142/FreeCAD-Gridfinity-Workbench #51: feature request: non-rectangular baseplates

kinghat, 2025-01-19:
ability to have any grid baseplate shape like an "L" so we can print as one piece and not have to glue parts together.

Stu142, 2025-01-19:
Your timing is perfect I just pushed an update with beta L shaped bins a couple days ago v0.8.0. L-shaped baseplates will be coming soon. 

I hope to introduce some other shapes as well such as T and U and would be open to suggestions. 

I think the idea of being able to make any custom shape is super cool, but I can't think of a way to implement that at this time with the way the workbench works.

kinghat, 2025-01-19:
even for circular shapes as well:

[image]

i was thinking about how a full custom baseplate thing would look like as well. i was thinking maybe setting your grid size for example 10x10 for max height x width your grid would go then a popup with that grid and you can visually select the cells you want the grid to populate with like this where the green would be the cells i selected:

[image]

i could open a new issue, but have you thought about baseplate connection points?

Stu142, 2025-01-19:
These are very interesting ideas. Circular would be cool and potentially doable. Selecting cells would be awesome but I am simply not familiar with how to even implement a pop up that would work like that with the workbench. I know it is possible with QT and macros so there might be a way I am just not familiar. It would be very complex to generate those shapes with so many possibilities, but it is probably possible. It would be far in the future, if anything, unless someone else wanted to tackle that. 

A new Issue would be good for the baseplate connection if you have a specific recommendation. Currently there is the screw together option and I would like to explore an alternative for snap together as you suggested but there are many things above that I want to add first.

greg19, 2025-03-18:
I believe this is now implemented and the issue can be closed

greg19, 2025-04-17:
@Stu142 is that correct?

Stu142, 2025-04-17:
Yes, I don't think circles will be implemented at this time.

closed 2025-04-17 by Stu142