Stu142/FreeCAD-Gridfinity-Workbench #56: Option to disable "Magnet Base Hole"

greg19, 2025-02-01:
When creating the "Magnet Baseplate", under the hole for the magnet there is another hole. What is its use case? Maybe there could be a parameter to disable it?

I didn't yet get around to glue the magnets to the plate, but my intuition is that it would be easier to do it without the holes?

Stu142, 2025-02-01:
I was thinking more from the perspective of making the magnets press fit and the hole as a way to remove them.

There is a parameter for adjusting the hole size in the Non standard section `Magnet Base Hole` and it seems to not give any errors if you set it to zero which gets rid of the whole. I don't think I considered the glue spilling out previously so It might be worth me mentioning this possibility in the documentation, thanks for bringing it up. 

If zero gives an error for you, try 0.001 mm or something, which makes it functionally zero, and there likely won't be a visible hole when printing.

greg19, 2025-02-01:
Thanks for the answer!

I didn't consider press fit (since [gridfinity.xyz/faq](https://gridfinity.xyz/faq/#question3) mentions only glue), but it seems like a great idea.

The issue for me is that a 6.5mm hole is just too big to hold a 6mm magnet in place. I printed one bin and plate from [gridfinitygenerator.com](https://gridfinitygenerator.com/) – it defaults to 6.2mm x 2.1mm holes. The magnet seems to fit tightly in the bin (removing it without through hole can be done somewhat easily), but – weirdly – it falls off the 6.2mm hole on the plate.

greg19, 2025-02-01:
Thanks for the answer!

I didn't consider press fit (since [gridfinity.xyz/faq](https://gridfinity.xyz/faq/#question3) mentions only glue), but it seems like a great idea.

The issue for me is that a 6.5mm hole is just too big to hold a 6mm magnet in place. I printed one bin and plate from [gridfinitygenerator.com](https://gridfinitygenerator.com/) – it defaults to 6.2mm x 2.1mm holes. The magnet seems to fit tightly in the bin (removing it without through hole can be done somewhat easily), but – weirdly – it falls off the 6.2mm hole on the plate.

If the defaults are set up for press fit, what is the rationale for 6.5mm hole? Maybe it could be lowered to 6.2, like in gridfinitygenerator?  
I guess one still needs to test the fit on a particular printer and magnet first (as my example with gridfinitygenerator shows). Maybe we could indicate it somewhere in this project?

Stu142, 2025-02-01:
6.5 is the default because that was on the original Gridfintiy bins from Zack. As you say, it's different for each printer, so some testing is required. I have read that 6.19 seems to work well and did for me as well. 

I will add a note about changing the value to work for your printer somewhere because that is a good idea. 

I honestly didn't put that much thought into the default, as it can just be changed to whatever you want, and everyone has a different objective and printer. You make a good point, and something like 6.2 mm as the default would be more appropriate. Thanks for your suggestions.

greg19, 2025-02-02:
By the way, I found a tooltip for `MagnetHolesShape`:
```
Shape of magnet holes. <br> <br> Hex meant to be press fit. <br> Round meant to be glued.
```
Considering what you are saying, I think it should be adjusted (saying that round are meant to be press fit too) – or left as is, and change the behavior so when you choose round, the base hole disappears by default.

The discoverability of all these documentation could be better, opening FreeCAD and hovering over every single parameter when there is this much of them is a pain (I didn't do it and didn't find this tooltip earlier...).

Stu142, 2025-02-02:
Yeah I agree with you there is a lot of parameters and its hard to tell what does what and how to interact with them. 

Better documentation is in order and there really should be an explanation and example of each feature to show what is does and how it can be manipulated.

To be completely honest I find it hard to find the motivation to make documentation as its much less exciting for me than adding new features for example but I do understand the importance and will work on that after this next big update I am currently working on.

closed 2025-02-07 by Stu142