Stu142/FreeCAD-Gridfinity-Workbench #57: Error when creating object: makeLoft() takes no keyword arguments

tree3589, 2025-02-01:
First time logging a defect, so please excuse lack of completeness etc - great plugin.

On update the v0.8 - get this error: Error when creating object: makeLoft() takes no keyword arguments

Am using both Ondsel 2024.2.2 and Freecad 0.21.2 on a MacBook - and see the issue in v0.8 in both applications. I rolled back to v0.72 in Ondsel by manually installing the zip into the Ondsel Mod folder, and that version works again.

Full error message from Freecad:
10:27:45  switching to Gridfinity Workbench
10:27:48  Traceback (most recent call last):
  File "/Users/___/Library/Application Support/FreeCAD/Mod/Gridfinity/./freecad/gridfinity_workbench/features.py", line 105, in execute
    gridfinity_shape = self.generate_gridfinity_shape(fp)
  File "/Users/___/Library/Application Support/FreeCAD/Mod/Gridfinity/./freecad/gridfinity_workbench/features.py", line 465, in generate_gridfinity_shape
    fuse_total = make_bin_base(obj)
  File "/Users/___/Library/Application Support/FreeCAD/Mod/Gridfinity/./freecad/gridfinity_workbench/feature_construction.py", line 679, in make_bin_base
    bottom_chamfer = Utils.rounded_rectangle_chamfer(
  File "/Users/____/Library/Application Support/FreeCAD/Mod/Gridfinity/./freecad/gridfinity_workbench/utils.py", line 157, in rounded_rectangle_chamfer
    return Part.makeLoft(wires, solid=True)
<class 'TypeError'>: makeLoft() takes no keyword arguments
10:27:48  Recompute failed!

Stu142, 2025-02-01:
Are you getting the error for everything or just a specific bin or specific combination of parameters.

Stu142, 2025-02-01:
I think the Gridfinity Workbench v0.8.0 is not compatible with that version of FreeCAD. 

I tried reinstalling the FreeCAD 21.2 and encountered a number of errors while trying to get the workbench to work. I am also on Windows and do not have access to MacOS to test. I would recommend updating to FreeCAD 1.0 as it is a large step forward in general. I will update incompatibility this on the documentation.

If you would rather not update you could try opening the Utils.py folder and changing line 157 to be: 

`return Part.makeLoft(wires, solid=True)`

into: 

`return Part.makeLoft(wires, True)`
This is how it was before and might be causing the problem but there are many other changes that could also be the cause of the errors.

tree3589, 2025-02-02:
Thanks for your response.

1 - I updated Freecad to the latest version and the issue is not observed anymore in the Mac version.
2 - I edited to the Utils.py as suggested above in the Ondsel installation, and that corrected the issue as well. 

I'm happy to test any updates on my MacBook as well.

closed 2025-02-04 by tree3589