Stu142/FreeCAD-Gridfinity-Workbench #64: Error when saving

Axolito, 2025-02-02:
Hello, I have version 0.8.1 of the workbench and FreeCAD 1.0.
I get the following error when I save:

```
22:31:18  PropertyPythonObject::toString(): failed for <class 'freecad.gridfinity_workbench.commands.ViewProviderGridfinity'>
22:31:18  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\json\encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\json\encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\json\encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```

At first I didn't have this problem, but then it appeared on a project I was working on. I tried with a new file and the error persists.
However, the error doesn't seem to interfere with saving the file.

Stu142, 2025-02-02:
Thanks for reporting, I happened to come across this today as well and I believe I have found the fix which is included in PR #55 which should be done within the week.

B48D81EFCC, 2025-02-06:
I have the same error while saving. I am looking forward to test the fix.

Stu142, 2025-02-07:
Fixed v0.9.0

kinghat, 2025-02-08:
im still getting this when saving:

> PropertyPythonObject::toString(): failed for <class 'freecad.gridfinity_workbench.commands.ViewProviderGridfinity'> (2 times)

> OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
> Architecture: x86_64
> Version: 1.1.0dev.40122 (Git) Conda AppImage
> Build type: Release
> Branch: main
> Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
> Logical/physical DPI: 96/157.74
> Installed mods: 
>   * OpenTheme 2024.9.1
>   * ThreadProfile 1.92.0
>   * fasteners.backup1737422417.4826305 0.5.35 (Disabled)
>   * sheetmetal.backup1737422500.1660883 0.7.4 (Disabled)
>   * sheetmetal 0.7.5
>   * fasteners.backup1738291620.497429 0.5.36 (Disabled)
>   * fasteners 0.5.37
>   * Gridfinity.backup1738814554.8302772 0.8.2 (Disabled)
>   * Gridfinity 0.9.1
>

Axolito, 2025-02-08:
After the update, I tested on a file created with the previous version and I still had the error. Then I tested by creating a new file with the new version and there I found no error.

In which of these two cases are you @kinghat ?

kinghat, 2025-02-08:
ya i just realized that you get the error with a previous version of the baseplate but if i remove that and use the newer version of baseplate i dont get the error.

Stu142, 2025-02-08:
I was just testing it myself and am able to save but getting an error when reopening:

> pyException: <string>(1)<class 'TypeError'>: FoundationGridfinity.loads() takes 1 positional argument but 2 were given

It doesn't appear to affect functionality. I know the fix and will have that out shortly.

closed 2025-02-09 by Stu142