Stu142/FreeCAD-Gridfinity-Workbench #70: scoop feature creates a wall wider than the rest when enabled

kinghat, 2025-02-09:
when scoop is enabled the wall on that side becomes wider than the rest of the walls:

[image]

[image]

Stu142, 2025-02-09:
This is an intentional feature based on the gridfinity originals. My guess is so small parts don't get caught on the stacking lip when retrieving them. 

I would be open to changing it based on recommendation. Additionally, I just checked and it stays thick even when the stacking lip is off, so that should definitely be changed, at least.

glenn20, 2026-01-06:
> I would be open to changing it based on recommendation. Additionally, I just checked and it stays thick even when the stacking lip is off, so that should definitely be changed, at least.

PR #135 fixes this as a side effect of adjusting the scoop location for the thin lip feature.

closed 2026-02-28 by Stu142