Stu142/FreeCAD-Gridfinity-Workbench #71: stacking lip not included in height total

kinghat, 2025-02-09:
i can sort of see why its not included in the total height when using height units but when using a custom height i think it should. usually the person is setting a specific height that the bin has to be.

[image]

also is it intentional to model the bin in the -Z from the origin vs above?

Stu142, 2025-02-09:
I see how that could be confusing for a user but at the same time I don't really see the stacking lip as part of the bin. If you were to stack another bin you would not be able to use that height. 

easy changes would be: adding to the tooltip when hovering that it doesn't include the label shelf. Or potentially a second reference dimension that does include it. 

I would be open to changing this if there was a really good proposed alternative. However, this would not be an easy change as this parameter is heavily used throughout both bin and baseplate generation code (but only bin would change) and I would want to have a super solid alternative before undertaking that.

hkienle, 2025-02-10:
> Or potentially a second reference dimension that does include it.

I think it would be useful to give the total height, width and depth of the bin. It would serve as a "sanity check" for the user. (You can get that information from the slicer, but it would be nice to have it in FreeCAD before generating the STL.)

hkienle, 2025-02-10:
> is it intentional to model the bin in the -Z from the origin vs above?

I guess this is most useful for blank (i.e. "filled") bins. This allows you to take away material starting from z=0 and going in the negative direction.

Stu142, 2025-02-10:
The consensus is clear then and I agree it does make intuitive sense that 'Total Height' actually means the total. 

@greg19 the reference parameters started as parameters I created for use in generating features then I thought they could be useful to the user so I made them visable as a result they are all quite heavily used throughout feature generation.

Stu142, 2025-02-10:
I suppose I could just do a find and replace for most of them so it shouldn't be too bad. 

Baseplates have 'TotalHeight' as well so I would have to make sure the change doesn't interfere.