Stu142/FreeCAD-Gridfinity-Workbench #76: Add label shelve by chosing a wall to attach.

greg19, 2025-02-10:
I want to propose an alternative method of adding label shelves or scoops. It would work by selecting a wall you want to attach the shelve to, then clicking a button on a toolbox:

https://github.com/user-attachments/assets/36b089f2-24fb-4c5f-a2e7-b4677a7ff6ff

I think it would be more user friendly that a pile of parameters, but only applicable if the user wants few shelves, as it isn't scalable when you have many dividers.

What could is nice in this approach is that the shelf is a separate object (which should appear as a child of the bin) that you can manipulate – so for example you don't have to rely on the hard-coded positioning options, but translate it in any way you want.

The code for this proof of concept isn't really interesting, but you can find it [here](https://github.com/Stu142/FreeCAD-Gridfinity-Workbench/commit/1fe81eee58de7c32c1d6b8b3d2323b17911ed42f).

This is somewhat related to #69.

Stu142, 2025-02-11:
This is a very cool idea. This would allow a lot of potential customization possibilities. 

Are you thinking of having this in addition to the already hard coded options? I like the idea of having hard-coded options as I see them as potentially more reliable and scalable.

greg19, 2025-02-11:
I think there are two main concerns when implementing label shelves (and scoops):
1. "Mass production" – for bins with many dividers and a shelf in each slot: it should be as easy as possible to manipulate all of them at once.
2. "Fine tuning": it should be flexible enough to create pretty much anything.

The current approach (where everything is a parameter or a bin object) is good in the first point, but not scalable in the second – making each possible modification a parameter is just not scalable (or even possible).

I think it would be possible to create a solution that solves both concerns well, but I lack the knowledge to come up with a good system. I think if it were to exist, it would:
- Have label shelves as separate CAD objects.
- The object should have a way to be cloned (but still be a single object) to every divider slot. When it is transformed, all of them should transform.
- Have some presets that could be set (I think mainly positioning: left/right aligned or centered, but maybe there is something else).
- Be a toolbox button that is selection aware. Maybe when a single face is selected, it would by default attach to it, and when a whole object is selected it would be more "mass production" oriented.

As for now, I think we are far away from such a system. A first step should be a implementation of this POC, but the current approach would still be necessary for mass production.

PR #83 (2025-03-16): Attachable label shelf
This closes #76.

[image]
[image]

There is some work that I want to do:
- [x] Make selection more robust
- [x] Standardize the initial height it appears on
- [x] Adding parameters to the object
- ~Making the object appear as a child of the bin in the tree~
- ~Making the object transform with the bin~
closed 2025-03-16 by Stu142