Stu142/FreeCAD-Gridfinity-Workbench #80: PartsBin object has no attribute bintype

kinghat, 2025-02-10:
im getting this when trying to enable the label shelf from "off" on my partsbin:

> PartsBin: 'PartsBin' object has no attribute 'bintype' (6 times)

```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/157.74
Installed mods: 
  * OpenTheme 2024.9.1
  * ThreadProfile 1.92.0
  * fasteners.backup1737422417.4826305 0.5.35 (Disabled)
  * sheetmetal.backup1737422500.1660883 0.7.4 (Disabled)
  * sheetmetal 0.7.10
  * fasteners.backup1738291620.497429 0.5.36 (Disabled)
  * fasteners 0.5.37
  * Gridfinity.backup1738814554.8302772 0.8.2 (Disabled)
  * Gridfinity 0.9.4
```

Stu142, 2025-02-11:
I was not able to reproduce the issue. Would you be able to confirm the version displayed in the bin properties as I find the addon manager can sometimes be finicky and there have been several updates recently. 

[image]

Is there more to the error message that you would be able to show. I believe I know where would be causing this but just want to confirm. 

Are there any other properties that you adjusted or a certain sequence of events that led to this. Are you able to reproduce with a specific sequence of events? Does this happen with any other bins?

kinghat, 2025-02-11:
> I was not able to reproduce the issue. Would you be able to confirm the version displayed in the bin properties as I find the addon manager can sometimes be finicky and there have been several updates recently.
> 
> [image]

looks like its lagging behind:

[image]

> Is there more to the error message that you would be able to show. I believe I know where would be causing this but just want to confirm.

nah that was it, directly copied from report view.

> Are there any other properties that you adjusted or a certain sequence of events that led to this. Are you able to reproduce with a specific sequence of events? Does this happen with any other bins?

not that i would know of, other than updating the wb. i only have the one bin.

Stu142, 2025-02-11:
I find the add-on manager update to be hit or miss but uninstalling and reinstalling is pretty reliable to get the latest version

kinghat, 2025-02-11:
> I find the add-on manager update to be hit or miss but uninstalling and reinstalling is pretty reliable to get the latest version

uninstalled the wb and restarted fc, installed the wb and restarted fc, opened file and it still showed the partbin at 9.1 version and get the same error when trying to enable the label shelf.

Stu142, 2025-02-11:
I will try and investigate further, I am still not able to reproduce the issue on any 0.9.x version. I haven't tried on the FreeCAD 1.1 dev version so there is a chance that could be it.

kinghat, 2025-02-11:
> I will try and investigate further, I am still not able to reproduce the issue on any 0.9.x version. I haven't tried on the FreeCAD 1.1 dev version so there is a chance that could be it.

if you can't repo with Dev version lmk and I can send you my file on discord or matrix freecad servers.

Stu142, 2025-02-11:
Is this using a previously saved bin or a newly generated. There have been pretty significant code structure changes, so the previous version might just not work if that is the case.

kinghat, 2025-02-11:
> Is this using a previously saved bin or a newly generated. There have been pretty significant code structure changes, so the previous version might just not work if that is the case.

its a 9.1 bin if thats what you mean? why are the objects versioned or why arent they upgraded as the wb upgrades?

Stu142, 2025-02-11:
The bins are versioned when created. A bin created on version 0.9.1 may not work on workbench version 0.9.4 because it calls the workbench to recompute, and in this case, something is missing. If you are opening a previously saved bin vs generating a new one this could be the case. 

I am still learning about what effects this has I had originally thought that more information was stored in the bin object itself but it seems it just calls the workbench with whatever version you have installed. However, the `bintype` property is added during initialization (generation), so it wasn't added to old bins. 

I didn't realize this change would have this effect. I have been working on more robust code and getting some guidance from some community members on how to do that, so hopefully, this can be improved in the future. for example I can make the property assigned in a section that is called for a recomputed so it would be added to old bins if they don't have them.

kinghat, 2025-02-11:
fwiw i dont get that error on newly created bins. feel free to close or do whatever you think is best for the issue.

Stu142, 2025-02-11:
I appreciate the report as I am still learning a lot, and in this case, compatibility could have been maintained if I had known, but I will know for next time.

Stu142, 2025-02-11:
I went back and tried opening a bin generated in 9.1 in workbench 9.4 and got the same error.

kinghat, 2025-02-18:
i get this with a duplicated 9.4 bin in 9.4 wb. probably still an enabled bin label thing 🤔 

[image]

[image]

closed 2025-04-27 by Stu142