Stu142/FreeCAD-Gridfinity-Workbench #85: Feature Request: Snap Connectors

collateral87, 2025-02-17:
Hi,
it would be nice, if the the snap connectors would find a way to this amazing Freecad addon.

Link to the files:
([https://www.printables.com/model/430144-gridfinity-base-with-snap-connectors/files](url)

kinghat, 2025-02-17:
> Hi, it would be nice, if the the snap connectors would find a way to this amazing Freecad addon.
> 
> Link to the files: [https://www.printables.com/model/430144-gridfinity-base-with-snap-connectors/files](url)

your link points to the wrong location.

i was going to create an issue as a repo or dump of all the variants to bases and bins for future reference. was also brought up here: https://github.com/Stu142/FreeCAD-Gridfinity-Workbench/issues/51#issuecomment-2600959709

> A new Issue would be good for the baseplate connection if you have a specific recommendation. Currently there is the screw together option and I would like to explore an alternative for snap together as you suggested but there are many things above that I want to add first.

here are the ones ive come across:

https://www.printables.com/model/452675-gridfinity-clickfinity-baseplate-no-magnets-univer

https://www.printables.com/model/719455-gridfinity-clickfinity-baseplate-w-connectors

https://www.printables.com/model/920601-gridfinityclickfinity-edge-baseplates-with-connect

https://www.printables.com/model/982173-clickbase-a-no-magnet-latching-gridfinity-baseplat

https://www.printables.com/model/413761-gridfinity-refined

https://www.printables.com/model/506105-clickfinity-refined-baseplates-for-use-with-gridfi

https://www.printables.com/model/718244-clickfinity-refined-lite

collateral87, 2025-02-18:
Thank you for your answer. I updated the url to the correct source.
In general this options would be very nice.

Stu142, 2025-02-22:
Good recommendation, and the license would allow use. There are several other big features in progress but I hope I could get to this eventually.

brummbrum, 2026-02-13:
The CLICKbase baseplate  by John Hall is my favourite ((also mentioned above, https://www.printables.com/model/982173-clickbase-a-no-magnet-latching-gridfinity-baseplat). Would be great to have a fully parametric version of it as part of this workbench's codebase.

I have created a generator template project in FreeCAD using stu's Gridfinity Workbench. It modifies the standard baseplate to replicate the CLICKbase features. It can create M x N sized grids with fully custom grid size (default is 42mm of course) and even fine tune the clamping force by tuning the thickness of the clamp blocks and the spring section of the frame. Just edit the variable set in the project and it generates your custom CLICKbase. You can download the project file here:

https://github.com/brummbrum/FreeCAD-Tools/blob/main/CLICKbase%20Template%20V2.FCStd

Same CC 4 license as John Hall's original files on printables. If you guys want to integrate CLICKbase natively into the workbench code the sketch dimensions and scaling  expressions should save you a lot of time as you could parse these dimensions into the corresponding Python calls.

I haven't bothered to add the plate-to-plate clips yet as I was mostly keen on holding bins in the grid without magnets. But those interconnections can be added easily too of course.

Hope this helps anybody who would like to get this useful variation of a baseplate into CAD and not having to mess around with stl meshes etc.