Stu142/FreeCAD-Gridfinity-Workbench #87: Baseplate perimeter fill to remove slop in drawers or boxes

IrataDev, 2025-02-20:
I am adding a fill perimeter around the baseplate to keep it stable within a drawer. Hopefully, I am not missing an existing feature, but I can't add this feature with the tools provided without doing more with creating negatives within FreeCad.

Possible solution:
I would like an option allowing the user to measure the internal x and y dimensions of the area where the baseplate is to reside. Once the outer perimeter is established, adding the ability to offset the grid in both the x and y dimensions within the perimeter allows better fitment by either a percentage or a measured distance. Furthermore, an option to radius or chamfer the lower edge of the outside perimeter and the corners would provide ease of use for fitment into non-perfect drawers or boxes.

greg19, 2025-02-22:
You can easily achieve it using built-in FreeCAD workbenches, for example using Part workbench:

1. Create your baseplate/bin in Gridfinity Workbench
2. Switch to *Part Workbench*
3. Use the *Measure* tool to check for your spacer dimension and location:
    [image]
    [image]
    For me, length is 76cm, height is 4.25cm, position is as shown
4. Create a Cube and give it proper position and dimensions
    [image]
5. Repeat for the other side if needed.
6. Optionally, you can combine the baseplate/bin with the spacers by selecting them both and using *Union* tool, but it isn't necessary if you are exporting to STL

You can play around with other tools to customize the spacer to your liking

Stu142, 2025-02-22:
As mentioned, there are ways to accomplish this in FreeCAD, but I am already planning to add this as a feature to baseplates as I do think it would be highly useful.