Stu142/FreeCAD-Gridfinity-Workbench #90: Make bin bases coincident with X & Y axes

nirvdrum, 2025-03-03:
I'm trying to use the "Bin Base" function of the Gridfinity workbench to serve as the base for a custom holder on top. I've been trying to make the exact shape so I can print a base that I'll be able to union cleanly. In this example, I need a 3x1 Gridfinity base. As far as I can tell, the width, height, and corner radius values are not accessible as VarSets or through a Spreadsheet. So, what I've done is created a new body and used the ShapeBinder combined with external geometry to get the bin base shape. Using some equality and coincident constraints, I have the exact shape I want.

To make the shape align exactly with the external geometry I've added construction points on my new shape and made the coincident with the border in the external geometry. While this works, I had to zoom in quite a bit because the bin base doesn't sit exactly on the X or Y axis. It's slightly offset, which makes it tricky to select. If it sat right on the X and Y I think it'd be overall easier to work with.

I've attached a screenshot showing the offset.

[image]

Stu142, 2025-03-03:
I am potentially open to changing this, but it is intentional and the reason for the current setup is the offset of the Clearance by the gridfinity standard of 0.25 mm on each side of a bin. This way was a little easier to code as every feature is centered on + 1/2 a grid unit. A baseplate does not have the clearance and would be flush and in this position if a baseplate was added the bin would be in the correct position relative to the baseplate. I have seen some people using that use case. 

I can also understand that most people wouldn't do that so It might be easier to have it flush. I will have to think on it before making any changes there. 

On another note you should be able to reference the bin properties in a spreadsheet for example `BinBase.xTotalWidth` or `BinBase.OuterBinRadius` will return the respective values as the are listed in the properties tab of the object. Better Documentation is desperately needed on this, but there have been some big updates recently and are still in the works, and I want to get those done first.

nirvdrum, 2025-03-03:
Thanks for the quick reply. The use case you outlined is interesting and I didn't realize the offset was intentional. If it's always 0.25mm, I can work with that. It's convenient being able to make the edges coincident to the axes, but it's not a requirement. The base was just so close to the axes that I assumed the intention was to make them coincidental. It looks like that's accessible as `BinBase.Clearance`, so I should be able to set the offset from the axes in my sketch. I tend to prefer making things coincidental just so I know they can't diverge, but if they're driven by the same attribute that should be fine.

Thanks for the guidance with the dimension references. I should've realized I could fetch them via `BinBase.<attr>`.

nirvdrum, 2025-03-04:
I found an even easier solution. I've adjusted the position of the Shape Binder to be `-BinBase.Clearance` for both the X and Y components. Then I don't need to deal with the outline of the shape being so close to the axes.