Stu142/FreeCAD-Gridfinity-Workbench #92: Add crush ribs for magnet holes

nirvdrum, 2025-03-03:
The [Gridfinity Rebuilt in OpenSCAD](https://github.com/kennetek/gridfinity-rebuilt-openscad) project added [support for crush ribs](https://github.com/kennetek/gridfinity-rebuilt-openscad/pull/187) for magnet holes and I've found it to be a great way to get magnets to stay in place without needing to glue them. The PR has a screenshot of what the feature looks like:

[image]

The picture lacks the center hole for pushing the magnet out, but that is an option. I don't think the Gridfinity workbench would need to add a solid fill option; the "Magnet Base Hole" parameter should be sufficient. Additionally, I've found the [default parameters](https://github.com/kennetek/gridfinity-rebuilt-openscad/blob/993814227204b942fc1c13e64a604427f835742a/src/core/standard.scad#L48-L53) for the rib size and count has worked remarkably well. I don't think that those would need to be configurable in the workbench, which should help the new shape fit in with the "Hex" and "Round" options currently available.

If you're not interested in expanding the magnet hole options, please feel to free to close this out. I'm not sure whether the current hex hole shape is meant for press-fitting magnets or to use small fasteners. If the former, I think the crush ribs may have an advantage in retaining the magnet by way of having more contact points.

I could try to pull this together, but I don't want to open a PR for a feature that the project isn't interested in.

Stu142, 2025-03-04:
I haven't seen this before, I think it would be a great addition to the workbench.

Stu142, 2025-04-11:
@greg19 Looks good. What are you thinking for default values? I like the look of the one with fewer ribs, but it might require testing. 

I agree that there would be an overlap with the hex, making it redundant. However, the hex was contributed to the project by someone else, so I don't want to be too quick to get rid of it. Thinking of keeping it for now, I don't think there is much downside of additional options with the way things are currently structured.

greg19, 2025-04-14:
I haven't tested any values in practice, and probably this is going to take me some time (as I don't have a printer at my place). If you want to test it yourself, I have incorporated these changes into #116.

I think we could make an issue to remove the Hex after merging this, and wait if anyone has any objections for some time.

Stu142, 2025-04-15:
I think that is a reasonable approach. I can print out some tests soon I am quite busy still for another week or so.