Stu142/FreeCAD-Gridfinity-Workbench #95: Exception: 'BinBase' object has no attribute 'bin_base_values'

nirvdrum, 2025-03-04:
I'm working on a design for a Gridfinity holder. I'm using the "Bin Base" feature of the Gridfinity workbench to form the basis of my design.

I found myself wanting a taller base so I tried to adjust the "Gridfinity->Height Units" option from `1` to `2` and right after that I started getting an exception that I've been unable to clear, even when restoring the value:

```
23:43:01  pyException: Traceback (most recent call last):
  File "/home/nirvdrum/snap/freecad/common/Mod/Gridfinity/./freecad/gridfinity_workbench/features.py", line 68, in execute
    gridfinity_shape = self.generate_gridfinity_shape(fp)
  File "/home/nirvdrum/snap/freecad/common/Mod/Gridfinity/./freecad/gridfinity_workbench/features.py", line 300, in generate_gridfinity_shape
    self.bin_base_values.make(obj)
<class 'AttributeError'>: 'BinBase' object has no attribute 'bin_base_values'
23:43:01  BinBase: 'BinBase' object has no attribute 'bin_base_values'
```

Gridfinity Workbench version: 0.10.2
FreeCAD version: 1.0.0
OS: Ubuntu 24.04

nirvdrum, 2025-03-04:
Okay. I played with this some more and modifying the value didn't really have an impact. I think it just forced an object recompute. The real problem is I had rebooted FreeCAD upgrading from v0.10.1 to v0.10.2 of the Gridfinity workbench. It looks like there's an issue persisting base bins. When reloaded, the `bin_base_values` attribute isn't restored.

A trivial reproduction is:

1. Create new document
2. Create new part
3. Create new body
4. Create new bin base
5. Save document
6. Close document
7. Re-open document
8. Run the _Recompute Object_ on the _BinBase_ object

nirvdrum, 2025-03-04:
I tried to `git bisect` this. I know I can recompute the object and update attributes in v0.10.0 and that it broke in v0.10.1. I think the immediate problem with persisting/loading/initializing `BinBase#bin_base_values` was introduced in eb5b3d68d798aa85894e427dfc909cdef7e7d602, but I'm having difficulty verifying because if I take the parent commit I get a different exception. I think that traces all the way back to 816526ec5dc67a5a37a3f4d07b908ddf73d79cf1, which is the first commit for PR #84.

I'm not really all that familiar with the FreeCAD Python integration, so apologies if I'm misattributing anything here.

@greg19 If you get a chance, can you please take a look at the reproduction and see if #84 may be the culprit?

greg19, 2025-03-04:
Ah, yes, this PR is the issue. That's why I was afraid to include the fixes while introducing the type checker.

The problem is that every `self.foo` have to be included in `dumps` and `loads` (which are currently neatly hidden in `FoundationGridfinity`) in order to be preserved between saves. Previously the `__init__` and `make` were used as standalone functions, not methods, and the `self.features` array was never used, so it wasn't a problem.

Tere are two solutions now:
- Serialize and deserialize them, but this would require (a) many copy pasting to serialize each feature independently (b) some aggregate approach, like `self.features` dictionary, which has it's own problems.
- Just make __init__ and `make` normal functions.

Coming back to my discussion with @Ruudjhuu in #84: 
> In my eyes all the features classes could be just pairs of functions – one that adds the parameters, second that constructs a shape. This is probably too extreme – wrapping them in a class can make the code structure cleaner (IMHO the simplicity of functions is also nice – even more for non experienced programmers – so this comes to personal preference). But I don't see why these classes should have a common superclass if they are always handled separately.

I think this is the only solution now, because *none* of the files saved currently have any features serialized – pairs of functions are the only way to maintain backwards compatibility. I will try to get a PR ASAP.

As a note, I'm working on some tests for saving the files (#96), and I will try to include this bug.

greg19, 2025-03-04:
> Are we confident that "don't use self.foo anymore in the future" Is the correct solution? It is a big language feature we are actively not using anymore.

We are using `self.foo` *a lot* through `addProperty` which adds a member and handles serialization.

Object proerties should be used to *store data* – that is what they are designed to do. This is not the case for Feature classes, as they are just simple namespaces (and artificially constrain signature) for pairs of functions. This is not the designed use case for classes, which are supposed mimic real life objects (that *do have* properties).

> We could make our classes dump-able and load-able. Which would eliminate the bug with saving and loading files when using the same workbench version, but will likely break backwards compatibility with any small changes we make.

The serialization of empty classes is just a noop. I think we had a discussion that the Feature classes should not own anything (have no members), because the object owns the properties. So this is the desired serialization. It is an even stronger argument that they are just pairs of functions, not a real objects.

We could obviously add a noop serialization, but this adds a lot of code that by definition does nothing and just complicates the architecture. This is undesired for both present and new developers (as it adds mental overhead) – but mainly for the latter, since it makes the code harder to understand (and adds a "why would someone do it this way" confusion for people that didn't participate in the discussion). Make things simpler, not more complex.

> It seems they pickle everything.

Did you check that they use pickle? It was my subconscious assumption, that broke #91 (it indicates they use JSON).

> Can't we somehow decouple the classes used by FreeCAD and the classes we are managing?

I don't see any benefits of this, but *a lot of* maintenance overhead. Once again, make things simpler, not more complex: _Pretend you were a really great version of PKZip, running continuously on your code, looking for ways to make it (semantically) smaller._ 

This is not what FreeCAD intends you to do. You should add a `obj.Proxy = self` to the object you created, and the proxy to operate on the object.

> Should I change my mindset more to functional programming, is that what FreeCAD intended?

FreeCAD wants you to use the DocumentObject as a source of truth, and operate on it. This is exactly what happens in both approaches, but in different manners.

NIT: It's [procedural programming](https://en.wikipedia.org/wiki/Procedural_programming), not [functional programming](https://en.wikipedia.org/wiki/Functional_programming).

greg19, 2025-03-04:
Also note that I'm not saying that we should *never* use members in proxy classes. Just use it only when it makes sense, like for example in custom bin layout ([features.py:161](https://github.com/greg19/FreeCAD-Gridfinity-Workbench/commit/4ab3965968ba7fe22b1c0b28cdd809dd3666e51c#diff-fd5f5d96cd8ddc07c9845747f2fc6b87864098fcd925b2e1183264a877de466aR161-R167) in new diff), that is constructed at runtime, thus it *has* to be serialized to be preserved between sessions.

PR #97 (2025-03-05): Make freatures pairs of functions
This fixes #95.

Features are now pairs of functions `foo_bar_properties` and `make_foo_bar`. This avoids any pitfalls during serialization. Sadly, GitHub is bad in diffing indentation :/.

Another bugfix in this PR is that custom shape layout is properly serialized when saving the file.
closed 2025-03-05 by Stu142