Stu142/FreeCAD-Gridfinity-Workbench #98: Removal of Lshaped bin

Ruudjhuu, 2025-03-04:
Now that we have the `CustomBin` (very nice work, thanks @greg19 and @Stu142) the `LShapedBin` seems to be obsolete. I think this should be removed. As I was writing some more tests, I think I'll skip code regarding the L bin for now. We can use this issue to track if someone is already working on it or not.

Stu142, 2025-03-04:
I'm not sure where I stand on this. The major functionality that is not replaced in the custom bin is the ability to change the grid dimensions and with that setup customizations that change with it. Which I think could be highly valuable but I also doubt there are many people using that. Additionally, I had planned to make inside corners independently adjustable so it could be sharp if the use case required it, but I didn't get to that yet. This would be challenging to replicate in the Custom bin as it would be needed to differentiate between inside and outside corners. 

looking forward, I want to add padding, which is adding a partial grid onto the side of a bin or baseplate. This would be much harder to add to the custom bin as there are many edge cases with the shapes. where with a known shape it would be easier to implement and made to be much more customizable such as a different padding for x1 and x2. 

What I have been currently envisioning is having the custom shape go alongside a couple of other common shapes, such as L and T and maybe U. While there is a lot of overlap, I see the strength is the amount of potential customization. None of this is really set in stone and will likely change just my current thoughts. I am thinking to keep it for now but I suppose it wouldn't be that hard to bring it back in the future if I wanted.

Stu142, 2025-03-04:
A potential fix to changing the dimensions could be if selecting the CustomBin command would edit the currently existing one. I believe I have seen functionality similar to this somewhere in FreeCAD but I don't remember what it was.

Stu142, 2025-03-05:
Okay, I agree. Get rid of the L bin and try to implement more functionality into the custom objects and see how far we can get with that. Some considerations for this approach. 

- the custom bin is slightly more computationally intensive due to the increased number of fuses, though I must admit it's not as bad as I expected. 
- There will need to be a method of dealing with the possibility of having multiple Custom bins in the same file for creation and changing dimensions.

closed 2025-04-15 by Stu142