Stu142/FreeCAD-Gridfinity-Workbench #101: LabelShelfWidth is inaccurate when no stacking lip

greg19, 2025-03-09:
[image]

The label shelf width is calculated as

```py
width = (
    obj.StackingLipTopChamfer
    + obj.StackingLipTopLedge
    + obj.StackingLipBottomChamfer
    + obj.LabelShelfWidth
    - obj.WallThickness
)
```

regardless if the object has a stacking lip or not. It can be misleading.

There is not really a good solution for this equation:
1. If we always include the stacking lip, a bin without it would have it inaccurate
2. If we never include the stacking lip, a bin with it would have it inaccurate
3. If we include the stacking lip only if it's enabled, it would make the shelf shrink/expand when the stacking lip is toggled

Currently the chosen option is 1. and I'm not saying it should be different *now* – it is probably more reasonable than the others, since a bin with a stacking lip is the "default". But I want to rethink this decision taking the standalone (attachable) label shelf into account. I want the standalone shelf to be truly the width `LabelShelfWidth`, but also it would be neat if creating a label shelf with these two ways acted exactly the same by default. There are two solutions for this:

- Use option 2., sadly this somewhat breaks backwards compatibility – old bins will change when recomputed. It could be solved with an `if obj.version > ...`, but this is a last resort in my opinion (as it adds heavy maintenance costs) and I would consider just breaking the compatibility first.
- Make the default width of standalone label shelf a bigger number to match the standard label shelf. This comes with a con of having standalone label shelf of width $W$ being a different shape that a normal label shelf of width $W$

I think the first solution is better.

Stu142, 2025-03-09:
By the first solution, do you mean: 

> Use option 2., sadly this somewhat breaks backwards compatibility – old bins will change when recomputed. It could be solved with an if obj.version > ..., but this is a last resort in my opinion (as it adds heavy maintenance costs) and I would consider just breaking the compatibility first.

This is a tough problem. I might be misunderstanding but I believe you are suggesting this would make the label shelf less wide than indicated in the properties if there is a stacking lip. I think there is potential for this to cause a bad experience if you print a bin with a label shelf stating a certain width and then you go to put on your labels and it is actually not wide enough.

greg19, 2025-03-12:
> I might be misunderstanding but I believe you are suggesting this would make the label shelf less wide than indicated in the properties if there is a stacking lip. I think there is potential for [...]

This was my idea, but now that I think of it you are probably right.

> I am not sure of an exact solution but I do think fairly strongly that the label shelf should be a minimum of the displayed value in all situations.

What do you mean by this?

Stu142, 2025-03-12:
> I am not sure of an exact solution but I do think fairly strongly that the label shelf should be a minimum of the displayed value in all situations.

I was basically reiterating what I said in the first paragraph. Label shelf width property should be equal to or less than the useable width of the label shelf to place a label. 

The built-in label shelves can be changed to have a different width depending on whether there is a stacking lip or not. For the attachable, this becomes tricky. A couple of thoughts I have had to potentially make this happen. 

- might be possible to determine if the object the stacking lip is attached to has the stacking lip on. 
  - Additionally a property of the individual label shelf could be added to indicate if it is on an outer edge or not. 
- pop up GUI when placing allowing selecting what type of edge there is that it is being placed on (stacking lip or divider). 
- a different command for edge label shelf vs internal. possibly confusing but not too bad. 
- detecting when placed if it is on an edge or internal. this become tricky with custom shapes.

greg19, 2025-03-13:
Thanks for the explanation. I don't really use the label shelf, so I don't want to decide if the width should change width dependent on the stacking lip.

Your ideas sound great, I will be thinking about how they can be implemented.

I was also thinking about just raising a popout with a message like
> Be aware that the stacking lip might cover the label shelf, making it less wide than expected. Measure the true width before printing.
[ ] Don't show again.
OK

if no other sensible option could be implemented.

Stu142, 2025-03-13:
That could be a pretty good solution as well. It may still be confusing to a new user who might not be sure how to check that distance or make the change.

closed 2026-02-28 by Stu142