Stu142/FreeCAD-Gridfinity-Workbench #106: Origin is not at the bin center for "Centered at Origin"

gnestlinger, 2025-03-16:
When I set the "Generation Location" option of a bin to "Centered at Origin", I would expect that the origin is at the actual center of the bin.
But, with the current version (0.10.6) it looks like this, where the center is offset from the origin in both the x- and y-direction:
[image]

Stu142, 2025-03-16:
Good catch, looks like I forgot to add the `Clearance` value to the location offset calculation when set to centered. Easy fix I'll have that out soon.

closed 2025-03-16 by Stu142

greg19, 2025-03-16:
@Stu142 I added tests for this to #96, and it's still a problem for BinBlank

greg19, 2025-03-16:
Yea, my bad

PR #108 (2025-03-16): fix bug
- Fix #106