Stu142/FreeCAD-Gridfinity-Workbench #109: Feature Addition: Changable Layout of Custom Objects

Stu142, 2025-03-16:
Putting this here to document my thoughts on the matter. 

This functionality would greatly boost the useability of the custom shape object. What I currently feel could be a nice way to do it is to have a separate command that will modify the layout of the object that is selected in the Tree. This would be a good alternative to repeating the generation command as it eliminates edge case scenarios where there are multiple custom objects or multiple of the same type. 

If it were possible to open the GUI with the current layout already selected, that would be pretty cool.

Ruudjhuu, 2025-03-28:
ohw, and we should probably adapt the constructor of the Dialog box to take a "begin grid_layout" as argument so the current layout is already painted. This `grid_layout` is just a two dimensional array of booleans.

greg19, 2025-03-29:
> I think if we "save" the layout information somehow in the `DocumentObject` (not sure which formats it accepts, two dimentional array would be great, serializable string as worst case scenario) and create an additional UI element with a command that:

It isn't necessary as the obj.Proxy.layout is already serialized so it can be just changed directly in the Proxy.

PR #112 (2025-04-05): Change layout command
This closes #109
closed 2025-04-05 by Stu142