Stu142/FreeCAD-Gridfinity-Workbench #118: Global Profil

ucool321, 2025-04-13:
Hi

I have a suggestion for the Gridfinity-Workbench:

Creating a "Global Profil".  Where there will be automatically overwritten Gridfinity Specs.

For example if I use the Gridfinity Spec but with some custom changes.

- Grid size
- Global disable magnet holes on the feet

etc.

greg19, 2025-04-14:
How do you envision the overwrite? Currently all bins have parameters that define the dimensions. I can imagine few ways this could behave:

1. The Global Profile affects only the values during the creation, like requested in #19. The work towards this is happening in #65, but it is a big feature so it's going slowly.
2. The global profile affects all bins in the document, having some parameters (like grid size or magnet/screw configuration), that would no longer be part of bin/baseplate parameters.
3. The global profile affects all bins in the document, but individual objects can override them. I don't know how this could be implemented in UX friendly way though – the solutions I can think of have really big drawbacks:
    - Some magic value (like 0) of the bin parameter would mean to use the Global Profile value, but could be confusing for users.
    - Each parameter would have an additional boolean flag "Use custom value", but this would really clutter the parameter list (which even now is too cluttered IMO).

Or are you thinking about something totally different?

ucool321, 2025-04-20:
Issue #19 and #65 will solve my idea.